nathalislight/NCAT #4: Esp32 wroom vs wrover

Babaika1987, 2022-06-21:
Здравствуйте какой из этих лучше использовать Esp32 wroom vs wrover? И если вам не сложно не могли бы вы написать гайд или видео с подробной инструкцией по прошивки и подключении вашего чуда.

Babaika1987, 2022-06-21:
> Wrover is really recommended, and wich version of NCAT?

Nescat 1.0

Babaika1987, 2022-06-21:
Я вижу 3 меню nes audio oscill, oscill работает два других нет при запуске
чёрный экран и не чего не происходит

BurAndBY, 2022-09-19:
> Nescat 1.0

Добрый день, проверял 0.5 на ESP WROOM и с маленькими играми проблем не испытывал

Были проблемы только с ROM файлами больше 1МБ. 

Если вам нужен емулятор NES лучше используйте 0.5 который был к сожалению удален с гитхаб

Заархивированая ссылка - https://github.com/markoni985/NesCat

BurAndBY, 2022-09-19:
Пару фото игры[image][image][image]

fizarum, 2023-01-22:
there are some forks left, for example: https://github.com/stivcen01/NesCat1.0_ESP32 (but its 1.0, not 0.5 as mentioned above)

BurAndBY, 2023-01-23:
> there are some forks left, for example: https://github.com/stivcen01/NesCat1.0_ESP32 (but its 1.0, not 0.5 as mentioned above), and that 1.0 not the same as 1.0 in this repo

My fork https://github.com/burandby/NesCat/ seems to be the only one out of two forks which has full files available