nathalislight/NCAT #7: Memory problem

DeJan-NS, 2022-09-03:
First off, I am a huge fan and I would love to build this NCAT system.
I wanted to flash the NCAT_PRO to the ESP32 WROVER and ran into some trouble.

The compiler says:

Arduino: 1.8.15 (Windows 10), Board: "ESP32 Wrover Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), QIO, 80MHz, 921600, None"

text section exceeds available space in board

**Sketch uses 2468390 bytes (188%) of program storage space.** Maximum is 1310720 bytes.

Global variables use 67588 bytes (20%) of dynamic memory, leaving 260092 bytes for local variables. Maximum is 327680 bytes.

Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

Error compiling for board ESP32 Wrover Module.

-Am I using incorrect settings for the wrover or does it take some tweaking the code to get it optimised? 
Even when I use the 8MB setting, I get the same error.
The only way it lets me compile is when I use the NCAT_MODULAR board (and even then it uses 96% of the memory). It doesn't even get to flashing the board, the compiler straight up tells me that there is no memory.

Thank you very much in advance!

eggfly, 2022-11-26:
> First off, I am a huge fan and I would love to build this NCAT system. I wanted to flash the NCAT_PRO to the ESP32 WROVER and ran into some trouble.
>
> The compiler says:
>
> Arduino: 1.8.15 (Windows [...]

Select dev board to "ESP32 Dev Module" and then we can see more options, including "Flash Size". Select an app size option with size more than 4MB.

ajsmith1624, 2023-05-07:
did you ever figure this code out to compile? if so let me know pleaseeee

ahmet58-robotik, 2025-11-07:
Did you figure out a way to compile