nathalislight/NCAT #9: Just a noob questions

skyrimXI, 2023-04-18:
1. what is the resistor value should use on buttons?
2. When clicking right button it seems work as button A as well as right what should i did wrong?

el-rom, 2023-12-04:
Hi ;
If you use externals resistors for buttons, you can use 4,7KOhm to 10KOhm value.
Don't forget to disable internal resistor (see INPUT.h include) but I seems that is the default.
To define exacts buttons pins connexions, check section HARDWARE CONFIGURATION & LIBRARIES in the main code and redefine input pins accordly to your own ncat system ... here an example for direct inputs buttons :

//--------------------------------------------------------------------------------
// KEY BUTTONS PINS: (Direct Buttons Pulleddown)
//--------------------------------------------------------------------------------
#if(GPIO_BUTTONS == 3)
//KEY BUTTONS PINS:
  #define PIN_UP          35  //
  #define PIN_DOWN   34  //
  #define PIN_LEFT       36  //
  #define PIN_RIGHT    39  //
  #define PIN_A      2  //
  #define PIN_B      4   //
  #define PIN_START   17  //
  #define PIN_SELECT 16  //
#endif

skyrimXI, 2023-12-06:
Thanks a lot

closed 2023-12-11 by skyrimXI