nathalislight/NCAT #11: Guru Meditation Error: Core 1 panic'ed

StefanAngelovski, 2023-09-22:
13:11:02.427 -> FLASH SIZE: 266256
13:11:02.427 -> Mapping 300000 (+41010)
13:11:02.427 -> mmap_res: handle=2 ptr=0x3f420000
13:11:02.427 -> [NES ROM MAPPED ON FLASH!]
13:11:02.427 -> Inserting cartridge.
13:11:02.427 -> Head:  78  69  83  26
13:11:02.427 -> ROMLOAD SUCCESS.
13:11:02.427 -> Inserting Cartridge...
13:11:02.427 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
13:11:02.427 -> Core 1 register dump:
13:11:02.427 -> PC      : 0x400da613  PS      : 0x00060530  A0      : 0x800df63d  A1      : 0x3ffb1f60  
13:11:02.460 -> A2      : 0x3ffd396c  A3      : 0x3ffd541c  A4      : 0x00000000  A5      : 0x3ffc1bc8  
13:11:02.460 -> A6      : 0x3ffbe584  A7      : 0x3ffe17dc  A8      : 0x3ffe35d8  A9      : 0x00000001  
13:11:02.460 -> A10     : 0x3ffe35d8  A11     : 0x3ffe359c  A12     : 0x3ffbe584  A13     : 0xdd3cc853  
13:11:02.460 -> A14     : 0x00000001  A15     : 0x3f420000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
13:11:02.460 -> EXCVADDR: 0x00000118  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
13:11:02.494 -> ELF file SHA256: 0000000000000000
13:11:02.494 -> 
13:11:02.494 -> Backtrace: 0x400da613:0x3ffb1f60 0x400df63a:0x3ffb1fb0 0x4008705e:0x3ffb1fd0
13:11:02.494 -> 
13:11:02.494 -> Rebooting...

How would I go by fixing this issue? I really want this to work but there seems to be some kind of issue reading from memory

evallgar, 2023-10-28:
This happened to me. Solved by changing in boards manager the esp32 version from 2.x to 1.0.5

scanlight, 2023-12-28:
@evallgar  and how to do this ? my board manager doesnt allow other cores then version 2