nathalislight/NCAT #12: Need support for st7735

Mainak009, 2023-11-08:
This is a very good project but that st7789/st7789v/st7789v2 isn't widely available in the market only st7735 is available with various display resolutions....so can you please make a compatible version(if possible and support) which can support st7735?....or guide me how to port this project for st7735(if possible)?....I have a 128x160 st7735 and a 4mb ESP32 wroom devkit where I can test and report back to you 😀

Mainak009, 2023-11-10:
Sorry about your health problems.
can you please guide me on which header to define and which line to change for changing to ST7735 from ST7789 in your code? I have some basic knowledge of the Arduino field but am new to this ESP field...I have v0.5 of your old deleted code....and here below is my 1.8" ST7735 Unit with red PCB and normal SD card reader.

[image]

[image]

Mainak009, 2023-11-23:
helloo...please reply🥺

el-rom, 2024-05-02:
Hi :) I like to dig up some old post ;)
You can watch at ~line 1000 into esp32-ncat.ino  :

<//--------------------------------------------------------------------------------
//  LCD INIT
//--------------------------------------------------------------------------------
#if LCD_ENABLED

#define LGFX_USE_V1
#include "src/LovyanGFX/src/LovyanGFX.hpp"
lgfx::Panel_ST7789 _panel_instance;
//lgfx::Panel_ILI9341 _panel_instance;
lgfx::Bus_SPI _bus_instance; // SPI>

Look at the #include : esp32-ncat/src/LovyanGFX/src/lgfx_user/LGFX_ESP32_sample.hpp
to find your lcd panel and define it : lgfx::Panel_ST7735S     _panel_instance;

If you make your custom board, dont forget to define NCAT_MODULAR and configure yours pins:
Example at ~line 1048 : 

<#elif defined(NCAT_MODULAR)        
        // SPI
        cfg.spi_host = HSPI_HOST;       //(VSPI_HOST or HSPI_HOST)
        cfg.spi_mode = 3;
        cfg.freq_write = LCD_SPI_SPEED; // 
        cfg.freq_read = 40000000;
        cfg.spi_3wire = false;
        cfg.use_lock = false;
        cfg.dma_channel = 1;
        cfg.pin_sclk = 14;
        cfg.pin_mosi = 13;
        cfg.pin_miso = -1;
        cfg.pin_dc = 26;>

In reading the code, you'll find all you need to make your own toy (bigs thx @nathalislight) but an ST7735 display have not enought pixels to render all necessary surface. You might have to use (minimum) an ST7789 with 240x240 resolution.

Have fun with diy ncat-system :)

[image]