nathalislight/NCAT #14: Problems  using SD card

scanlight, 2023-12-26:
Can compile for the esp32 wrover

but cant use the SD card
E (1085) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1085) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1085) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1091) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1132) gpio: gpio_set_level(226): GPIO output gpio_num error

scanlight, 2023-12-26:
Full error after reboot.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
E (1085) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1085) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1085) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1091) gpio: gpio_set_level(226): GPIO output gpio_num error
E (1132) gpio: gpio_set_level(226): GPIO output gpio_num error
SD error!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

gad969, 2024-03-18:
I have same problem. Did you solve it?

scanlight, 2024-03-19:
No never got it to work. Sadly

gad969, 2024-03-19:
> No never got it to work. Sadly

I try old version of nescat yesterday and it works. https://github.com/markoni985/NesCat
It use other pins for sdcard module

Eques12, 2025-04-11:
> > No never got it to work. Sadly
> 
> I try old version of nescat yesterday and it works. https://github.com/markoni985/NesCat It use other pins for sdcard module

Ive been trying to get this to work but it keeps saying the SD card library is wrong and that video composite out cannot be defined, may i ask how you did it?