nathalislight/NCAT #15: Arduino IDE compilation error

gadiscz, 2024-05-15:
Hi, I am not able to compile in Arduino Ide. I choose ESP32 Wrover Module but I get a lot of errors every time. I tried different versions of both the ESP board and other versions of Nescat (0.5, 1.0).

I have a Nescat Modular NEW board.

Maxmoon56, 2024-05-15:
If you got error while compiling then it's not the board, maybe some missing library or some part of the code is not working, what's the error you get?

gadiscz, 2024-05-16:
So I tried again and it compiled. (Probably wrong fw version) Here I have a problem with the SD card. SD says it's OK but doesn't see any files (I have NES and AUDIO in the root) inside I have contra.nes and xxx.mp3

closed 2024-05-21 by gadiscz