nathalislight/NCAT #18: compiler error

namachoco0228, 2024-08-02:
I executed NCAT_ESP32.ino in NCAT_ESP32, but C:\Users\PC_User\Downloads\NCAT-main\NCAT_ESP32\NCAT_ESP32.ino:652:10: fatal error: soc /rtc_wdt.h: No such file or directory
  652 | #include "soc/rtc_wdt.h"
      | ^~~~~~~~~~~~~~~
compilation terminated.
Compilation error: soc/rtc_wdt.h

Compilation error: soc/rtc_wdt.h: No such file or directory is displayed and cannot upload. What should I do?

MecanixCreations, 2024-08-09:
same issue here

madmarcel, 2024-09-28:
From a comment I found elsewhere:
"Hi, I have the same issue, I use ESP32 and need roll back to 2.0.17. Hope it can help you."

Downgrade this board package to 2.0.17, code compiles after that.
[image]